tetrabiodistributed/project-tetra #29: Off the shelf parts

Burhan-Q, 2020-05-16:
Collect list of part numbers, specs, links, and documents for any/all off the shelf parts used. These will be listed in the final BOM and assembly documentation, for others to reference if looking to recreate the splitter system.

Burhan-Q, 2020-06-22:
### Tetra valve
v4-1

[G-74 - Compression Spring](https://www.centuryspring.com/catalog/?page=product&cid=root&id=G-74CS)
   Spring critical specifications:
spring rate = 0.385 N/mm (2.2 lbs/inch )
**minimum** inner diameter = 4.44 mm (0.175")
**max** solid length = 2.3 mm (0.09")

[O-Ring: -018 N70 NITRILE (NBR, BUNA) 70 DUROMETER BLACK](https://www.globaloring.com/product/n70018/)
[O-Ring: -020 N70 NITRILE (NBR, BUNA) 70 DUROMETER BLACK](https://www.globaloring.com/product/n70020/)
[O-Ring: -128 N70 NITRILE (NBR, BUNA) 70 DUROMETER BLACK](https://www.globaloring.com/product/n70128/)

v5
[G-74 - Compression Spring](https://www.centuryspring.com/catalog/?page=product&cid=root&id=G-74CS)
   Spring critical specifications:
spring rate = 0.385 N/mm (2.2 lbs/inch )
**minimum** inner diameter = 4.44 mm (0.175")
**max** solid length = 2.3 mm (0.09")

[O-Ring: -018 N70 NITRILE (NBR, BUNA) 70 DUROMETER BLACK](https://www.globaloring.com/product/n70018/)
[O-Ring: -020 N70 NITRILE (NBR, BUNA) 70 DUROMETER BLACK](https://www.globaloring.com/product/n70020/)

Burhan-Q, 2020-06-22:
### Shutoff valve

[4x O-Ring: -361 N75A Viton Fluoroelastomer](https://www.mcmaster.com/9464K623)
[2x Zinc-plated steel M4 x0.7 hex nut](https://www.mcmaster.com/90591A141)
[2x 18-8 stainless steel M4 x 0.7 screw 15 mm](https://www.mcmaster.com/92095A125)

Burhan-Q, 2020-06-29:
**Computer and Display**

[7" Touch display](https://www.raspberrypi.org/products/raspberry-pi-touch-display/)
[Raspberry Pi (additionally appropriate power supply, cables, storage, and other basic accessories](https://www.raspberrypi.org/products/raspberry-pi-4-model-b/)
_Notice: RPi4 in use for prototype, but at the moment no known limit for selecting a model; recommend RPi3 or newer_

capsulecorplab, 2020-07-07:
Linking Hannah's [Hardware parts list](https://docs.google.com/spreadsheets/d/1nCw9LUN7ygv8B8hno6poN-m15rsixNSolVTXhXynrfw/edit?usp=sharing) for cross-reference.