tetrabiodistributed/project-tetra #64: Add clearance to RPi display frame

Burhan-Q, 2020-07-20:
Add 0.25 mm to current width on inside RPi display frame

cyunis, 2020-07-23:
video with description of issue
[20200720_141340.mp4.zip](https://github.com/tetrabiodistributed/project-tetra/files/4963838/20200720_141340.mp4.zip)

Burhan-Q, 2020-07-23:
@mechatroniks-git What material are you using to print this? The original dimensions for the part are shown in the image:
[image]

I will add an additional 0.25mm as you mentioned, but would like to track what kind of material shrinkage we're seeing to avoid future issues.

Burhan-Q, 2020-07-23:
@mechatroniks-git change has been made, ready for review. Going to move card to under review on project board.

mechatroniks-git, 2020-07-24:
@mmroden printed it with PLA

mechatroniks-git, 2020-07-24:
Since we already have a -0 STL in github, we need to bump the part number to a -1. 

Dimension change appears to be good.
[image]

Burhan-Q, 2020-07-24:
Understood. I will update part text, but @mechatroniks-git, when I save the new part should I create a new milestone for Rev 1 and update the part number to reflect the revision?

mechatroniks-git, 2020-07-24:
I'd only save a milestone whenever you upload the file to git

Burhan-Q, 2020-07-24:
Agreed (with a minor caveat). I think we have a running discussion of this topic [here](https://helpfulengineering.slack.com/archives/G012Q55KWN7/p1595612022005500) about that. I think once we've hammered out the details, I'll be working with @cyunis to formalize it into the GitHub Docs.

mmroden, 2020-07-27:
less than a day to update, 2 days review

mmroden, 2020-07-27:
to be printed by Mark, reviewed by Patrick

mmroden, 2020-07-27:
will be printed in white

Burhan-Q, 2020-07-28:
@mmroden ready for printing
[P-TET-000004-1.zip](https://github.com/tetrabiodistributed/project-tetra/files/4989315/P-TET-000004-1.zip)

Burhan-Q, 2020-07-29:
@mmroden printed with a nice, uneven marbling; will reprint

[image]

mechatroniks-git, 2020-08-07:
Ready to test part when handoff is made

Burhan-Q, 2020-08-07:
> Ready to test part when handoff is made

@mmroden will you be able to get this part to @mechatroniks-git before/by next week? If not, recommend we move this ticket to Blocked (if there is no reply, I will move the ticket after 24 hours)

Burhan-Q, 2020-08-17:
@mmroden delivered to @mechatroniks-git this past weekend. Assigning to @mechatroniks-git for review/approval when available.

mechatroniks-git, 2020-08-26:
We will test this during assembly in the next few days

commit b06d9ab7 (2020-08-31): Design files Alpha sprint 2

closes #64
closes #67
closes #96
closes #95
closes #115
PR #128 (2020-09-01): Alpha sprint p2
closes #64 
closes #67 
closes #96 
closes #115

Accidentally duplicated previous PR inclusion of drawing template files, no changes should have been made
commit de3b20d1 (2020-09-01): Alpha sprint p2 (#128)

* Create folder

* Issue #98

* Issue #97

* Clean up release vs testing files

Add Archive folder for previous revision models
Add Testing folder for models not released but ready to test
Migrated For_Printing folder with currently released models

P-TET-000019-2 Issue #98
P-TET-000021-1 Issue #97
P-TET-000022-1 Issue #99
P-TET-000023-1 Issue #100
P-TET-000028-0 New iteration of Tetra valve with steps added to outlet

* Updated P-TET-000030 to -1

* Design files Alpha sprint 2

closes #64
closes #67
closes #96
closes #95
closes #115
closed 2020-09-01 by Burhan-Q