tetrabiodistributed/project-tetra #100: P-TET-000023 Design Updates

mechatroniks-git, 2020-07-21:
- [ ] Method to constrain nuts did not prevent them from spinning during assembly
- [ ] 22M port is not ISO-5356 compliant
- [ ] DFA - Remove extra one way valve nub if not needed to avoid assembly confusion
- [ ] DFAM - Darryl had recommendations for changes

cyunis, 2020-07-23:
Video showing failure to constrain nut from 7/20:
[20200720_173240.mp4.zip](https://github.com/tetrabiodistributed/project-tetra/files/4963913/20200720_173240.mp4.zip)

mmroden, 2020-07-27:
Time estimate: 2 days + 2 days revision

jgrano, 2020-07-30:
@mechatroniks-git 22M port is not ISO-5356 compliant same issue as on the shutoff valve.

jgrano, 2020-07-30:
https://github.com/tetrabiodistributed/project-tetra/blob/master/02_Development/Prototype_release/Alpha/Parts_to_print/P-TET-000023-1.stl - Normal

Burhan-Q, 2020-08-01:
As of today. @DHHwangPhD printed part[ P-TET-000028](https://github.com/tetrabiodistributed/project-tetra/blob/master/02_Development/Prototype_release/Alpha/Parts_to_print/P-TET-000028-0.stl) and at present is curing. Will deliver to @mechatroniks-git as soon as possible

Burhan-Q, 2020-08-05:
@mechatroniks-git if you could review P-TET-000028 (this is the version with the outlet steps) when you have a chance for the ISO fittings, I can export the STL to @DHHwangPhD for test printing once I get the green light from you

Burhan-Q, 2020-08-05:
Tetra valve female coupler (P-TET-000028 version) located in Tetra Valve Fusion folder

Opening diameter = 22.4mm
Length = 21mm (minimum spec)
End taper diameter = 21.8mm
Taper angle ~ 1.637 deg [full]

Did not remove sharp edge as I'm unsure if it will be an issue. Added fillets to aid transition regions

mechatroniks-git, 2020-08-05:
@Burhan-Q [image]
Opening diameter is 22.37, not 22.40
Taper is 1.43deg = atan(1/40), not 1.637deg 
Here is ISO Spec: https://www.iso.org/obp/ui#iso:std:iso:5356:-1:ed-4:v1:en

mechatroniks-git, 2020-08-05:
On this one, the taper is too short. The diameter at 15mm is correct (22.37mm), so need to extend the taper back another 6.5mm
[image]

Also, fillet this edge so it's not sharp
[image]

mechatroniks-git, 2020-08-05:
taper angle is wrong
[image]

mechatroniks-git, 2020-08-05:
Nuts are 7mm nominal, measuring 6.9mm, and the opening is 7.5mm, I'd reduce it to 7.1mm
[image]

Burhan-Q, 2020-08-06:
Tetra valve updates. Not sure how to redo outlet valve as single part without breaking geometry.

[image]
[image]
[image]

mechatroniks-git, 2020-08-06:
@Burhan-Q P-TET-000028 tweaked & reviewed, ready for test print.

Burhan-Q, 2020-08-06:
@DHHwangPhD ready for print testing. Please comment when you've started printing. 

[P-TET-000028-1.zip](https://github.com/tetrabiodistributed/project-tetra/files/5037885/P-TET-000028-1.zip)

Judoguy12, 2020-08-24:
@jgrano Can you check if the heat inserts and oring's are added to the tetra valve bit, as they are not showing up in the animation, Jamie

Burhan-Q, 2020-08-26:
> @jgrano Can you check if the heat inserts and oring's are added to the tetra valve bit, as they are not showing up in the animation, Jamie

@Judoguy12 worst case you can check out the parts list [here](https://docs.google.com/spreadsheets/d/1nCw9LUN7ygv8B8hno6poN-m15rsixNSolVTXhXynrfw/edit#gid=80304938&range=B18), find the correct size parts, and then add them into the model.

Judoguy12, 2020-08-28:
Do you have an idea of what o-ring it is on the middle part?

Burhan-Q, 2020-08-28:
> Do you have an idea of what o-ring it is on the middle part?

@Judoguy12 I looked at the model (A-TET-001030-0), and seems like it has an o-ring in it, PN V-MCM-9262K525

[image]

Judoguy12, 2021-02-10:
@mechatroniks-git I believed this is solved, the worked fine for my assembly