tetrabiodistributed/project-tetra #103: A-PET-000009 - Design Updates

mechatroniks-git, 2020-07-21:
- [ ] Open up holes to allow ISO5356 22M port to clear
- [ ] Add part # to model

Judoguy12, 2020-07-22:
@mechatroniks-git Where is this part located?

mechatroniks-git, 2020-07-22:
In the main assembly, it's the 4 dual port covers where the inspiratory/expiratory connections are

Judoguy12, 2020-07-23:
Updated and ready for review @mechatroniks-git

mechatroniks-git, 2020-07-24:
@Judoguy12 The hole is still too small. 
[image]

Here is the ISO Spec 
[image]

Judoguy12, 2020-07-24:
Done @mechatroniks-git

mmroden, 2020-07-27:
Patrick will review, John will approve

Burhan-Q, 2020-07-28:
PEER check - Port diameters are 22.75mm and have enough space between to allow for 2" filters

closed 2020-07-31 by mechatroniks-git