tetrabiodistributed/project-tetra #104: P-TET-000018 - Design Updates

mechatroniks-git, 2020-07-21:


jgrano, 2020-07-30:
@mechatroniks-git Can you send me some more close up pictures of the sensirion and the mating problems on this thread?

jgrano, 2020-07-30:
@mechatroniks-git ISO5356 fitting is the same as we used originally on the Flow sensor model that you sent me. If there are dimensional errors, it's likely printing.

mechatroniks-git, 2020-07-31:
Comparing the sensirion 22M to the shutoff valve (P-TET-00018-0) port, with a 22F/22F coupler, the sensirion fits snugly and seals. The valve body bottoms out without sealing
[image]

mechatroniks-git, 2020-07-31:
Measurements:
[image]

mechatroniks-git, 2020-07-31:
@jgrano Checking the Sensirion model, it appears to match the ISO Spec pretty well. So the part I'm measuring might have some shrinkage from the injection molding process. I remeasured the sensor with my better calipers, got 22.29mm

[image]

Burhan-Q, 2020-08-01:
As of today. @DHHwangPhD printed part and is curing. Will deliver to @mechatroniks-git as soon as possible

Burhan-Q, 2020-08-04:
@mechatroniks-git and/or @jgrano please review update to Tetra valve female coupler (P-TET-000028 version) located in Tetra Valve Fusion folder

Opening diameter = 22.4mm
Length = 21mm (minimum spec)
End taper diameter = 21.8mm
Taper angle ~ 1.637 deg [full]

Did not remove sharp edge as I'm unsure if it will be an issue. Added fillets to aid transition regions

Burhan-Q, 2020-08-05:
Added revolve on both ports to conform to ISO standard with 1.432-degree taper angle from 22.2mm to 22.37mm over 15mm

@jgrano and/or @mechatroniks-git please review design and give 👍 if I'm okay to send this to @DHHwangPhD for test printing.

[image]

Burhan-Q, 2020-08-05:
Feedback from @mechatroniks-git on this part

https://github.com/tetrabiodistributed/project-tetra/issues/100#issuecomment-669461211
https://github.com/tetrabiodistributed/project-tetra/issues/100#issuecomment-669462750
https://github.com/tetrabiodistributed/project-tetra/issues/100#issuecomment-669500081

Burhan-Q, 2020-08-05:
> On this one, the taper is too short. The diameter at 15mm is correct (22.37mm), so need to extend the taper back another 6.5mm
> [image]

Per the ISO spec, the 22.37mm diameter dimension for the male insert is to be at 15mm (unless I'm misreading the spec or unclear on the change needed here). Additionally, your comment from earlier on this Issue shows the same distance measurement. 
[image]

Previous comment from July 31
> Measurements:
> [image]

Burhan-Q, 2020-08-05:
> taper angle is wrong
> [image]

@mechatroniks-git it looks like the measurement point is in the wrong spot
[image]

When I measure to the outside, I get 0.707-degrees which is still not half of 1.432-degrees (1.414-degrees), but it's quite close and again I was following the instructions you laid out above. I can reduce the outer radius to 10.99mm to get 1.49-degrees of taper if that's preferable. 
[image]

Burhan-Q, 2020-08-06:
Shutoff valve updates
Taper now extends the entire length of ports
[image]

Nut capture openings now reduced to 7.1mm and added 0.3mm fillets
[image]

mechatroniks-git, 2020-08-07:
@Burhan-Q unless we have a good reason, we should strive to meet the ISO spec with a 1.432 deg taper, instead of 1.414.

Burhan-Q, 2020-08-07:
> @Burhan-Q unless we have a good reason, we should strive to meet the ISO spec with a 1.432 deg taper, instead of 1.414.

@mechatroniks-git I have no good rationale for keeping the larger angle. I've been focused the design updates primarily from the diameters for the locations, but since we last spoke I am more aware that the standard is more concerned with the taper than diameters. I will make this adjustment

Burhan-Q, 2020-08-07:
Design updated (v32) for port design to be based on taper angle, feature mirrored to other port. 

[image]
[image]

mechatroniks-git, 2020-08-08:
@Burhan-Q Looks great, I think we should print one

Burhan-Q, 2020-08-12:
@DHHwangPhD, version ready for test printing

[P-TET-000018-2.zip](https://github.com/tetrabiodistributed/project-tetra/files/5065343/P-TET-000018-2.zip)

Burhan-Q, 2020-08-12:
Reassigned for test printing. After printing complete, assignment follows whomever has the physical part or goes to anyone agreeing to make design changes (if needed).