tetrabiodistributed/project-tetra #105: P-TET-000006 - Design Updates

mechatroniks-git, 2020-07-21:
- [ ] ISO5356 22F port too small to mate with M-SEN-SFM3300 22M
- [ ] O-Ring groove in slip port too small, could not install o-ring
- [ ] Slip port is too loose, (27mm opening to 26mm tube)
- [ ] Tap NPT hole full depth
- [ ] Emboss part number in model

mechatroniks-git, 2020-07-23:
[image]

jgrano, 2020-07-24:
@mechatroniks-git Fusion still lacks this capability as a FYI for the NPT threads. The thread profile was made by using the MC part as a cutting tool.

jgrano, 2020-07-24:
@Judoguy12 made a quick change out the outer diameter of the ISO section of this part to increase wall thickness. Will print. The part was pulled from the main assembly and is now residing in \tubing in the fusion teams project as a fork.

jgrano, 2020-07-24:
@mechatroniks-git How do you feel about changing the NPT port to a straight threaded M3 and using push to connect for the air hose? 

https://www.mcmaster.com/5225K519/

Judoguy12, 2020-07-24:
Please do that it is much easier

mechatroniks-git, 2020-07-24:
For this build, let's stick with parts on hand (time & cost). For next iteration, I don't know if M3 in FDM/SLA would be robust with the leverage that can be exerted during tubing insertion, much better for an aluminum part.

mechatroniks-git, 2020-07-24:
@Burhan-Q had designed this part, maybe the threads were added in Solidworks, not Fusion?

mmroden, 2020-07-27:
John will do it, Patrick will review, Jamie does final approval

mechatroniks-git, 2020-07-27:
@jgrano - for the NPT hole, if you can add a chamfer, it helps with tapping and installing 

[image]

jgrano, 2020-07-29:
Printed out changes, Internal ISO fitting still feels a bit loose.

Judoguy12, 2020-07-29:
@jgrano Shoud be tighter now, if you were referring to the 26 dismeter hole

jgrano, 2020-07-29:
@mechatroniks-git Chamfer isn't coming out super clean on the npt section. Did some work around but still not super happy with it. 

In addition I decreased the diameter of the ISO internal again since it is pushing too far along the shutoff/tetra valve ports before locking in place.  Printing again 7/29/20

jgrano, 2020-07-30:
Printing again 7/30/20

jgrano, 2020-08-01:
Had a print fail last night. Starting again today.

Burhan-Q, 2020-08-11:
@mechatroniks-git and @jgrano, just checking in on this issue. I'm seeing a mismatch in Part Number and Part Name on the title of this issue. According to the [hardware doc](https://docs.google.com/spreadsheets/d/1nCw9LUN7ygv8B8hno6poN-m15rsixNSolVTXhXynrfw/edit#gid=0&range=B6) P-TET-000005 is the Pressure Manifold (SLA) and P-TET-000006 is the ISO adapter.

I didn't want to make any changes to this issue before verifying what part is being worked on/discussed here. It appears there are notes for both parts in the above comments, but I'm also not seeing a separate issue for each of these parts. Please let me know what comments are associated with which part and then I'd be happy to separate the issues.