tetrabiodistributed/project-tetra #67: Connector for Sensirion SFM3300-D

fpdotmonkey, 2020-07-21:


fpdotmonkey, 2020-07-21:
https://www.mill-max.com/products/new/rugged-4mm-pitch-spring-loaded-connectors

fpdotmonkey, 2020-07-21:
https://www.nxp.com/products/interfaces/ic-spi-serial-interface-devices/ic-bus-repeaters-hubs-extenders/i2c-bus-extender:P82B715

mechatroniks-git, 2020-07-28:
Attempted to use https://www.digikey.com/product-detail/en/avx-corporation/009155004201006/478-4684-1-ND/1687183
Contacts slip off the pad when compressed. 
[image]

mechatroniks-git, 2020-07-28:
This Samtec part is better, but interferes: https://www.digikey.com/product-detail/en/samtec-inc/SIB-104-02-F-S-LC/SAM10715-ND/2664704

[image]

mechatroniks-git, 2020-07-29:
Going with this part should work: https://www.digikey.com/products/en?keywords=821-22-004-10-002101

[image]

fpdotmonkey, 2020-07-29:
Seems like this tab doesnt extend out far enough[image]

Burhan-Q, 2020-07-29:
@fpdotmonkey Just curious, did you try clipping it from both directions? Happy to extend that tab, let me know what Δ you're seeing as needed.

[image]

mechatroniks-git, 2020-07-29:
@hydronics2, here is the PCB DXF
[Sensirion PCB.zip](https://github.com/tetrabiodistributed/project-tetra/files/4996695/Sensirion.PCB.zip)

mechatroniks-git, 2020-07-29:
The step file: 
[Sensirion PCB Assembly Mockup.zip](https://github.com/tetrabiodistributed/project-tetra/files/4996702/Sensirion.PCB.Assembly.Mockup.zip)

fpdotmonkey, 2020-07-29:
@Burhan-Q looking at it closer, I think the issue is actually the fillet on the inside corner of the part stopping it from fully engaging.  That said, it is clipping on and staying affixed. [image]

fpdotmonkey, 2020-07-29:
It could also be a print quality issue.

Burhan-Q, 2020-07-29:
@fpdotmonkey I can make that a sharp corner if you'd like to check for a better fit. Let me know what you'd like to do.

fpdotmonkey, 2020-07-29:
Sure sounds good

fpdotmonkey, 2020-07-29:
Also what's the part number on this?

hydronics2, 2020-07-30:
@mechatroniks-git  you're cutting it kinda close here. I'd recommend moving the connector 1-2 mm outboard or just trimming the PCB 1-2mm. Any input?
[image]

hydronics2, 2020-07-30:
this is just a general note to remember for later... The P82B715 level shifts the i2c signals to 5V.
If we end up putting the 2nd chip on the master board, 
be sure it level shifts back down to 3.3V or take other action as needed.

hydronics2, 2020-07-30:
@mechatroniks-git  do we need jumpers? I figure either the P82B715 is landed or it's not. Cheap small boards easy to switch between those with the P82B715 and those without. right? I can put them on no sweat just thought I'd simplify things as needed.

Burhan-Q, 2020-07-30:
> Also what's the part number on this?

@fpdotmonkey just now logged it on the [Hardware Parts List](https://docs.google.com/spreadsheets/d/1nCw9LUN7ygv8B8hno6poN-m15rsixNSolVTXhXynrfw/edit#gid=0&range=B29) as P-TET-000027

hydronics2, 2020-07-30:
@mechatroniks-git  for review
[image]

mechatroniks-git, 2020-07-31:
@hydronics2 
Can we add a parallel cap on the IC? Datasheet calls for .1 and 1uF. Also, add a pullup on SDA1. Thanks

[image]

hydronics2, 2020-07-31:
@mechatroniks-git I made the changes... why only a pull-up on the SDA line and not an SCL line? 
[image]

hydronics2, 2020-07-31:
@mechatroniks-git @fpdotmonkey @Burhan-Q  I submitted a pull-request. Here are the files. https://github.com/hydronics2/project-tetra/tree/master/02_Development/FlowMeter/Sensirion_connector_board
I estimated the pogo pin distance in the step file. The pogo pins stick out 5mm (0.197") from the board and the stroke is 1.39mm (0.055").  
[image]

fpdotmonkey, 2020-08-01:
@Burhan-Q @hydronics2 ping me on slack when you have the board and the clip ready for me to print

hydronics2, 2020-08-01:
> @Burhan-Q @hydronics2 ping me on slack when you have the board and the clip ready for me to print

I posted a step file tin the link above. Check it out and see what you think.

fpdotmonkey, 2020-08-01:
@hydronics2 in application, are the header pins in the model you posted going to be replaced by the sprung pins like what Patrick suggested?

hydronics2, 2020-08-01:
Yes that's right. I put some header pins in their and tried to size them to 5mm to represent the pogo pins that will stick out 5mm from the board according tonl the data sheet I referenced (uncompressed)

fpdotmonkey, 2020-08-01:
@Burhan-Q so I tried filing out that fillet, and that didn't do anything.  However, I bent it a little so the internal angle on it is a bit smaller, and now the fit is a bit better.

fpdotmonkey, 2020-08-01:
It basically looks like the cross section picture you sent me, so I think it's good.  Time to add a feature for the pins to go through I suppose.

Burhan-Q, 2020-08-01:
Moved to Blocked. See notes [here](https://github.com/tetrabiodistributed/project-tetra/pull/111#issuecomment-667583686).
PCB revision required to address issue shown in image

[image]

hydronics2, 2020-08-01:
haha.. that's funny. I didn't move that part I swear.... but I can certainly move it outboard as many mm as necessary... say start with 8mm?

mechatroniks-git, 2020-08-03:
We have a mystery to solve, last I left it, the connectors were 8.43mm (center to center distance from the row).  Pushing it out 5mm would work, give me the center to center distance, and I can check.

[image]

Burhan-Q, 2020-08-03:
Perhaps it's the spacing from the front edge of the board to the phoenix connector thru holes? [both measurements captured in Fusion, from 'front face' of board to center of thru hole]

@mechatroniks-git board measurement **{M}**
[image]

@hydronics2  board measurement **{H}**
[image]

Profile view with phoenix connector visible, looks like there's a difference in phoenix connector position/length. 

@mechatroniks-git **{M}**
[image]

@hydronics2 **{H}**
[image]

Burhan-Q, 2020-08-03:
> @Burhan-Q so I tried filing out that fillet, and that didn't do anything. However, I bent it a little so the internal angle on it is a bit smaller, and now the fit is a bit better.

@fpdotmonkey do you mean that I should decrease the angle of the clip section like this?
[image]

hydronics2, 2020-08-03:
@mechatroniks-git @Burhan-Q  So it looks like the pogo pins are fine but the large connector exact position needs to be resolved. I can put it wherever it needs to be. This is the latest revision... 
[image]

Burhan-Q, 2020-08-03:
Blocked by #113 

- [ ] adjust position of phoenix connector pins

fpdotmonkey, 2020-08-04:
@Burhan-Q the design for the clip is good no need to change it

Burhan-Q, 2020-08-05:
@fpdotmonkey clip updated with cut out for PCB. Ready for testing

[P-TET-000027-0.zip](https://github.com/tetrabiodistributed/project-tetra/files/5029886/P-TET-000027-0.zip)
[image]

Burhan-Q, 2020-08-05:
> @fpdotmonkey clip updated with cut out for PCB. Ready for testing

@fpdotmonkey please move to card to Testing bucket on project board and add Testing Label to the Issue once you begin printing.

fpdotmonkey, 2020-08-05:
First print of the clip didnt turn out so well. The hook that grabs the ledge nearest the PCB broke off while i was removing support material. Im going to try printing it with the PCB face down to see if that helps at all[image]

fpdotmonkey, 2020-08-05:
This is how I'm orienting the next print
[image]

fpdotmonkey, 2020-08-05:
The previous print was just like this one, but on the other outer face of the clip

fpdotmonkey, 2020-08-05:
Ok so the connector is nominally working!! The clip connects to the sensor perfectly, and the PCB slides right in. That said, i feel like its really easy to break the PCB-side hook while removing support, and the features for retaining the PCB didnt resolve super well in the print. [image]

fpdotmonkey, 2020-08-05:
Also, the clip has a good bit of slop along the sensor's axis (~.075").  I'm concerned that this could cause a short between pins.

fpdotmonkey, 2020-08-05:
It also only takes a hair of pressure to remove the PCB.  Granted we'll see how hard it is to inert and remove when we have the actual board, but I think an iteration

Burhan-Q, 2020-08-05:
> Also, the clip has a good bit of slop along the sensor's axis (~.075"). I'm concerned that this could cause a short between pins.

@fpdotmonkey could you get me two photos of the extremes of the motion you're seeing, with the Sensirion facing the other direction? I'd looking to see how the harness is moving relative to this feature
[image]

fpdotmonkey, 2020-08-05:
Full right ![Uploading 89474819...]
Full left ![Uploading 89474827...]

fpdotmonkey, 2020-08-05:
You can most easily see the difference looking at the right edge of the clip.  I measured the displacement to about .075"

Burhan-Q, 2020-08-05:
What's the inside distance you measure on the Sensirion here?
[image]

hydronics2, 2020-08-06:
@fpdotmonkey @Burhan-Q  should I order the PCB? 

Also Fletcher, just checking but you have the mating phoenix phoenix connector... maybe do a quick visual inspection to make sure it plugs in; that is,  the female connector can mate with the PCB-male-phoenix connector. make sense?

fpdotmonkey, 2020-08-06:
@hydronics2 yes order the PCB

fpdotmonkey, 2020-08-06:
@Burhan-Q that distance is .727"

hydronics2, 2020-08-06:
OK, PCB is ordered. thanks, thomas
Don't forget to order the pogo pins.
https://www.digikey.com/product-detail/en/mill-max-manufacturing-corp/821-22-004-10-002101/ED8199-04-ND/1146641

Burhan-Q, 2020-08-06:
> @Burhan-Q that distance is .727" or 18.47mm

> You can most easily see the difference looking at the right edge of the clip. I measured the displacement to about .075"

So I can tighten this up some if need be. I didn't know the exact spacing so I left 0.9mm clearance overall (which is about half the amount of movement you're seeing). The contact pads (on the Sensirion model) are 2.3 mm wide and have 0.24 mm gap between them, so I'll lower the spacing to 0.5mm and that should avoid any chance of a short between pins.

### Sensirion SFM3300-D model ### 
[image]

### Clip current width ### 
[image]

fpdotmonkey, 2020-08-07:
For the print, I used Hatchbox white PLA.  The spool is a few years old; idk if the material quality degrades over time.

For settings, I had the temps set to 225 end/80 plate, the "standard quality" template on Cura, and the retract speed set to 45mm/s

If there's a way to save out the entire settings file, I'd be happy to do that, but a quick search didn't yield me that result.

Burhan-Q, 2020-08-07:
> For the print, I used Hatchbox white PLA. The spool is a few years old; idk if the material quality degrades over time.
> 
I believe it will if not humidity controlled 

> For settings, I had the temps set to 225 end/80 plate, the "standard quality" template on Cura, and the retract speed set to 45mm/s
> 

Those temps seem quite high (to me), I also use Hatchbox PLA and print at 200/50 w/ default retraction speed

> If there's a way to save out the entire settings file, I'd be happy to do that, but a quick search didn't yield me that result.

Would be one of the 2 CFG files found in this directory (Win10), not all settings are included, my settings for the print are shown at the bottom. You can also pull the settings from the GCODE file as well, I made a [PowerShell script](https://github.com/Burhan-Q/PowerShell_projects/blob/master/read_PrusaSlicer_gcode.ps1) that pulls the settings from a Prusa Slicer GCODE files (which should be mostly similar to the Cura files).

%APPDATA%\cura\

```
[general]
version = 4
name = creality_base_extruder_0 #2_user
definition = creality_ender3

[values]
brim_width = 2
filter_out_tiny_gaps = True
infill_enable_travel_optimization = True
infill_sparse_density = 15
ironing_only_highest_layer = True
retraction_hop_enabled = True
skirt_line_count = 2
speed_infill = 75
speed_print = 60
support_connect_zigzags = False
support_infill_rate = 10
support_tower_maximum_supported_diameter = 1
support_tree_angle = 60
support_tree_branch_diameter = 1
support_tree_branch_distance = 0.01
support_tree_collision_resolution = 0.1
support_wall_count = 0
wall_line_count = 4
wall_overhang_angle = 70
wall_overhang_speed_factor = 80
z_seam_type = sharpest_corner
```
**Additionally:**
adaptive_layers = True
layer_height = 0.2
slicing_tolerance = Middle

fpdotmonkey, 2020-08-11:
Got the board in, but it's wider than we expected. The one I 3D printed is 18.08mm wide, whereas this one is 18.88mm. We'll need to adjust the clip to handle this as presently it doesn't fit at all. @Burhan-Q [image]

fpdotmonkey, 2020-08-11:
In case it's relevant, the fully extended length of the pogo pins from the top surface of the board is 6.79 mm and the fully retracted length is 5.76mm

fpdotmonkey, 2020-08-11:
The big green Pheonix connector is 16.14mm wide, 9.21mm deep, and 7.06mm above the top surface of the board

fpdotmonkey, 2020-08-11:
Here's a drawing of the dimensions of the board from the factory + me soldering the pogo pins and pheonix connector on
[Sensirion Connector Board.pdf](https://github.com/tetrabiodistributed/project-tetra/files/5059935/Sensirion.Connector.Board.pdf)

Burhan-Q, 2020-08-12:
@fpdotmonkey could you get me the inside dimensions (HxWxD) from the printed harness (height to under the hold downs)? The current model has an inside width of 19.80mm, so it seems there's some significant material shrinkage and knowing what the difference is will be instructive on how to modify the dimensions.

The model I printed measures:
   depth ~ 20.2mm
   height ~1.5mm (overall pcb harness height minus bottom of pcb harness to bottom of hold downs)
   width ~19.5mm

[image]

[12 more replies]

commit b06d9ab7 (2020-08-31): Design files Alpha sprint 2

closes #64
closes #67
closes #96
closes #95
closes #115
PR #128 (2020-09-01): Alpha sprint p2
closes #64 
closes #67 
closes #96 
closes #115

Accidentally duplicated previous PR inclusion of drawing template files, no changes should have been made
commit de3b20d1 (2020-09-01): Alpha sprint p2 (#128)

* Create folder

* Issue #98

* Issue #97

* Clean up release vs testing files

Add Archive folder for previous revision models
Add Testing folder for models not released but ready to test
Migrated For_Printing folder with currently released models

P-TET-000019-2 Issue #98
P-TET-000021-1 Issue #97
P-TET-000022-1 Issue #99
P-TET-000023-1 Issue #100
P-TET-000028-0 New iteration of Tetra valve with steps added to outlet

* Updated P-TET-000030 to -1

* Design files Alpha sprint 2

closes #64
closes #67
closes #96
closes #95
closes #115
closed 2020-09-01 by Burhan-Q