tetrabiodistributed/project-tetra #76: ⇑ Tetra Valve with One Way Exploded

Burhan-Q, 2020-07-21:
**Create exploded view:** A-TET-001010-0 Tetra Valve with One Way

Judoguy12, 2020-07-21:
21/07/2020 @mechatroniks-git

Burhan-Q, 2020-07-24:
I'm not 100% on this, but I would think that since the o-ring is installed from the bottom side, you'd want to move it under that section of the valve. I say this since I think the idea of the drawings is to demonstrate how this would be assembled, so the parts should be positioned relative to how they'll be installed. Otherwise, I think this looks good

[image]

Judoguy12, 2020-07-24:
Did that not update? i had moved that 😓

Judoguy12, 2020-07-24:
@Burhan-Q Should be fixed now

Burhan-Q, 2020-07-24:
Exploded view looks ready for drawing #77 

Closing issue

closed 2020-07-24 by Burhan-Q