tetrabiodistributed/project-tetra #85: Standalone Shut Off valve Exploded

Burhan-Q, 2020-07-21:
**Create exploded view:** A-TET-001015-0 Standalone Shut Off valve

Judoguy12, 2020-07-22:
22/07/2020 @mechatroniks-git

Judoguy12, 2020-07-24:
@Burhan-Q Ready for review

Burhan-Q, 2020-07-24:
When first opened Exploded view, components were hidden
[image]

Burhan-Q, 2020-07-24:
@Judoguy12 
- [ ] move blue highlighted parts up some 
- [ ] P-TET-000020-1 needs to be clear of base
- [ ] try to space o-rings evenly
- [ ] remove leader line from nut (unless it can be shown as inserting into catch

[image]

Judoguy12, 2020-07-24:
@Burhan-Q All fixed now

mmroden, 2020-07-27:
1 day estimate

Burhan-Q, 2020-07-28:
@Judoguy12 Exploded view looks good, ready for approval @mechatroniks-git