tetrabiodistributed/project-tetra #87: ⇑ Expiratory Support mounts Exploded

Burhan-Q, 2020-07-21:
**Create exploded view:** A-TET-001014-0 Expiratory Support mounts

Judoguy12, 2020-07-22:
22/07/2020 @mechatroniks-git to review

Judoguy12, 2020-07-24:
@Burhan-Q Ready for review

Burhan-Q, 2020-07-24:
@Judoguy12, please point me where to find this exploded view, I'm unable to locate it. Should only need to comment with the Directory and Filename, no direct link needed.

Judoguy12, 2020-07-24:
@Burhan-Q Animations tab of the prototype v1 system, storyboard with the largest number

mmroden, 2020-07-27:
Burhan will review, John will do final approval

Burhan-Q, 2020-07-28:
@jgrano review changes look good, ready for approving

Burhan-Q, 2020-07-28:
@Judoguy12 consider making the following changes:
- [ ] adjust view so left edge is at (or nearly) 90-degrees
- [ ] **OPTIONAL** add leader lines for the screws
[image]

Burhan-Q, 2020-07-28:
@Judoguy12  Exploded view looks good. @jgrano ready for approval

mechatroniks-git, 2020-07-29:
@Burhan-Q Missing the heated inserts (x 4)

Burhan-Q, 2020-07-30:
@Judoguy12 can you add the heat-serts and comment when updated? I believe it would be [these](https://www.mcmaster.com/94180A353/), but check the hole depth. You can find the off-the-shelf parts list [here ](https://docs.google.com/spreadsheets/d/1nCw9LUN7ygv8B8hno6poN-m15rsixNSolVTXhXynrfw/edit#gid=80304938&range=A1)if you need it

Judoguy12, 2020-07-30:
Done @Burhan-Q

mechatroniks-git, 2020-08-05:
Can we make these parts the same? Also, the diameter of each of these is different?

[image]

mechatroniks-git, 2020-08-07:
Fusion search for 001014 not coming up with any results, need to pull this subassembly out of the main assembly

Burhan-Q, 2020-08-25:
Reassigned to @Judoguy12 per discussion from meeting on 2020-08-24

Drawing appears to be ready for approval. @Judoguy12 will need to find an approver and assign the task to the approver (exclusively), once approver has provided feedback or approved, the assignment returns to @Judoguy12 until documentation is ready for pull request. All prior feedback is still applicable for the work to be considered complete and ready for approval.

Judoguy12, 2020-08-26:
@Burhan-Q Ready to approve