tetrabiodistributed/project-tetra #95: Create Drawing Template

Burhan-Q, 2020-07-21:
- [x] Assigned _(specify assignees)_
- [x] Work in Progress _(move card on Project Board)_
- [ ] Submit for Review _(comment w/ date and assigned reviewer)_
- [ ] Working on Review changes _(comment w/ list of changes and date)_
- [ ] Final Review complete _(assigned reviewer comment with approved date)_

Burhan-Q, 2020-07-21:
Status update: awaiting logo design from @thehannahfink and @cyunis for adding to template

Burhan-Q, 2020-07-21:
@jgrano Assigned as Reviewer

mechatroniks-git, 2020-07-23:
Drawing templates needed (mostly just notes changes)
- FDM Part Template (perhaps make it 2 sheets, 2nd sheet for heated inserts/post processing)
- SLA Part Template
- Assembly Template

Notes on Assembly Template:
```
Notes (unless otherwise specified):
1. This drawing is construed as information only, it is complete only to the degree necessary to illustrate a system definition for assembly

```

mechatroniks-git, 2020-07-31:
@Judoguy12 I updated the template
- Replaced logo with a image with transparent background I created. (@thehannahfink, do you have a set of PNGs with high resolution and transparent backgrounds?)
- Added units, tolerances and 3rd angle projection box

todo:
Look at replacing the font with Helvitica Bold?
update notes
add open source license info?

Judoguy12, 2020-08-03:
Ready for review @jgrano 3/8/2020

mechatroniks-git, 2020-08-03:
Title block needs help, logo looks rough, @thehannahfink can we do SVG art? [Fusion ](https://knowledge.autodesk.com/support/fusion-360/learn-explore/caas/sfdcarticles/sfdcarticles/How-to-import-SVG-files-in-Fusion-360.html)allows it to be added. 
[image]

Copyright notice - @capsulecorplab does this look right?

Judoguy12, 2020-08-03:
@mechatroniks-git Fusion would not allow the SVG to be added I had to convert to transparent PNG

Judoguy12, 2020-08-03:
@mechatroniks-git fixed address appearing

capsulecorplab, 2020-08-04:
@mechatroniks-git copyright notice looks good!

mechatroniks-git, 2020-08-05:
@Judoguy12 we need to place our web address somewhere

Judoguy12, 2020-08-05:
@mechatroniks-git Is that better or shall i remove the company name above the web address
[image]

mechatroniks-git, 2020-08-05:
The TETRA BIO DISTRIBUTED is redundant and doesn't match the style guidelines, so best to remove that
The unnecessary blank-space above & below the logo/name should be eliminated.
Maybe swap the logo with the website or combine them in 1 box?

Judoguy12, 2020-08-06:
iS THIS BETTER? 
[image]

thehannahfink, 2020-08-06:
[There is a version on the Git with just the logo and no title](https://github.com/tetrabiodistributed/tetrabiodistributed.github.io/blob/create-logo-6/assets/logo/black/tetra_logo_black_100_x_100_px.svg) if that's the logo version you want. The website should be all on one line, so maybe make the font size small enough to fit without a line break. I liked the website better on far-right column. Should be underneath the Project block in its own "website" block though.

commit b06d9ab7 (2020-08-31): Design files Alpha sprint 2

closes #64
closes #67
closes #96
closes #95
closes #115
PR #127 (2020-08-31): Closes #95: Drawing templates
Files created by @mechatroniks-git for use as drawing templates for use with Alpha prototype

Closes #95
commit de3b20d1 (2020-09-01): Alpha sprint p2 (#128)

* Create folder

* Issue #98

* Issue #97

* Clean up release vs testing files

Add Archive folder for previous revision models
Add Testing folder for models not released but ready to test
Migrated For_Printing folder with currently released models

P-TET-000019-2 Issue #98
P-TET-000021-1 Issue #97
P-TET-000022-1 Issue #99
P-TET-000023-1 Issue #100
P-TET-000028-0 New iteration of Tetra valve with steps added to outlet

* Updated P-TET-000030 to -1

* Design files Alpha sprint 2

closes #64
closes #67
closes #96
closes #95
closes #115
closed 2020-08-31 by Burhan-Q