tetrabiodistributed/project-tetra #96: A-TET-001005 - Add wires to assembly

mechatroniks-git, 2020-07-21:
Need to add power and ground wires to the assembly. These typically come with the RPi display, but they need to be added as a component (virtual or modeled) and noted they need to be modified

Burhan-Q, 2020-07-23:
Added virtual (empty) components to assembly, moving ticket to Review

mechatroniks-git, 2020-07-24:
Try to use the existing nomenclature to avoid confusion during assembly:
[image]

[image]

Burhan-Q, 2020-07-24:
@mechatroniks-git, Understood I will make these changes. The wires appeared to be grounding wires in the photos/videos posted but I did not look up they're purpose, and thought with respect to the BOM the wires would not need color specification.

Once changes are complete, should the Revision Number for the assembly be updated due to the addition of the virtual components?

mechatroniks-git, 2020-07-24:
This would be a drawing revision from - to A. The assembly didn't change, just a documentation change.

Burhan-Q, 2020-07-24:
Understood. Changes have been made. Please close if approved

mmroden, 2020-07-27:
Jamie will do final approval

Judoguy12, 2020-07-28:
Approved 28/07/2020

Burhan-Q, 2020-08-05:
Issue can be closed when #80 is complete and ready for pull request

commit b06d9ab7 (2020-08-31): Design files Alpha sprint 2

closes #64
closes #67
closes #96
closes #95
closes #115
PR #128 (2020-09-01): Alpha sprint p2
closes #64 
closes #67 
closes #96 
closes #115

Accidentally duplicated previous PR inclusion of drawing template files, no changes should have been made
commit de3b20d1 (2020-09-01): Alpha sprint p2 (#128)

* Create folder

* Issue #98

* Issue #97

* Clean up release vs testing files

Add Archive folder for previous revision models
Add Testing folder for models not released but ready to test
Migrated For_Printing folder with currently released models

P-TET-000019-2 Issue #98
P-TET-000021-1 Issue #97
P-TET-000022-1 Issue #99
P-TET-000023-1 Issue #100
P-TET-000028-0 New iteration of Tetra valve with steps added to outlet

* Updated P-TET-000030 to -1

* Design files Alpha sprint 2

closes #64
closes #67
closes #96
closes #95
closes #115
closed 2020-09-01 by Burhan-Q