tetrabiodistributed/project-tetra #97: P-TET-000021 Design Updates

mechatroniks-git, 2020-07-21:
- [ ]  Thin wall section can't print, needs to be cut back or thickend up. 
- [ ]  Loose fit when mating to P-TET-000019
- [ ]  Needs more clearance when mating to P-TET-000020
- [ ]  Consider DFAM tweaks

cyunis, 2020-07-23:
Videos from 7/20 showing screw details mostly and modifications to improve fit:
(also see Issue #98 for more videos of part P-TET-000021 with P-TET-000019 fit)

[20200720_181008.mp4.zip](https://github.com/tetrabiodistributed/project-tetra/files/4963976/20200720_181008.mp4.zip)

mechatroniks-git, 2020-07-23:
[image]

jgrano, 2020-07-23:
[image]
0.25mm spacing between the knob and the inner stub on part 20

mmroden, 2020-07-27:
1 day + 2 days revisions

jgrano, 2020-07-29:
@mechatroniks-git We're ready to have you print this.

jgrano, 2020-07-30:
https://github.com/tetrabiodistributed/project-tetra/blob/master/02_Development/Prototype_release/Alpha/Parts_to_print/P-TET-000021-1.stl

mechatroniks-git, 2020-07-31:
Printing

mechatroniks-git, 2020-08-04:
The knob appears to fit to P-TET-000019 Shutoff Valve Thru Body pretty well. Need to adjust P-TET-000020 to allow for clearance when spinning.

closed 2020-08-04 by mechatroniks-git