tetrabiodistributed/project-tetra #98: P-TET-000019 Design Fixes

mechatroniks-git, 2020-07-21:
- [ ] O-ring grooves too tight
- [ ] OD fit with SLA tetra valve was ok, but with FDM shutoff valve, it was a bit tight
- [ ] Pocket to hold nut needs to be changed so that nut doesn't spin
- [ ] Consider modifying pocket to hold nut such that nut kept centered during assembly (drop in to shallow pocket?)

cyunis, 2020-07-23:
Videos from 7/20:
[Archive.zip](https://github.com/tetrabiodistributed/project-tetra/files/4963966/Archive.zip)

mechatroniks-git, 2020-07-23:
[image]

jgrano, 2020-07-23:
Took all above notes into consideration rework of this part. Will reprint 7/23 and verify fit w/o o-rings

Judoguy12, 2020-07-23:
John, when you save the part you could put in the reason for saving the
issue number so it is traceable

jgrano, 2020-07-23:
[image]
@Judoguy12 so add #98 to the final save?

jgrano, 2020-07-23:
Final save of design and starting to print based on all feedback.

Burhan-Q, 2020-07-27:
Darryl to print for testing

jgrano, 2020-07-29:
Body that I printed works great, integrates fine with existing shutoff valve body.  Waiting for verification from Darryl

mechatroniks-git, 2020-07-29:
John, nice work. Can I go ahead and print the FDM parts on this?  should I pull STLs from Fusion? Point me in the right direction, want to make sure I print the right ones. Thanks.

jgrano, 2020-07-31:
https://github.com/tetrabiodistributed/project-tetra/blob/master/02_Development/Prototype_release/Alpha/Parts_to_print/P-TET-000019-2.stl

Burhan-Q, 2020-07-31:
> https://github.com/tetrabiodistributed/project-tetra/blob/master/02_Development/Prototype_release/Alpha/Parts_to_print/P-TET-000019-2.stl

@DHHwangPhD ready for testing

mechatroniks-git, 2020-08-04:
Part fits knob much better, nut capture is good also. O-ring compression is too much. Need to revise. Currently it is a 2mm o-ring compressed 35%. We should try 10% compression. See https://www.marcorubber.com/o-ring-groove-design-considerations.htm

[image]

mechatroniks-git, 2020-08-05:
Reduced o-ring groove ID such that o-ring compression is 10%, not 35%. Should make it easier (possible) to assemble.

Bumped part number embossed on part to P-TET-000019-3

[image]

Ready to print & test

mechatroniks-git, 2020-08-07:
design updated to -3, ready for review @Burhan-Q

Burhan-Q, 2020-08-07:
> Part fits knob much better, nut capture is good also. O-ring compression is too much. Need to revise. Currently it is a 2mm o-ring compressed 35%. We should try 10% compression. See https://www.marcorubber.com/o-ring-groove-design-considerations.htm
> 
> [image]
> 
> Also, not sure we can do much about it, but the thin section at the bottom, but as the groove goes deeper, this will be weaker.

[image]
I'm seeing the change as 1.90 mm (5% of 2mm o-ring). If this is still okay, then I think this part looks good.

General note for main body vs thru body nut capture dimensions, request was to reduce main body nut capture to 7.1 mm [here](https://github.com/tetrabiodistributed/project-tetra/issues/100#issuecomment-669500081), but stated above that this size (7.3 mm) was good.

> Part fits knob much better, nut capture is good also.

Burhan-Q, 2020-08-07:
@jgrano @Judoguy12 
Ready for approval for whoever is available. Please self assign this issue if you're able to Approve

mechatroniks-git, 2020-08-07:
@Burhan-Q good catch on the oring groove depth. I'm ok to test it with 5% compression on first article

Burhan-Q, 2020-08-17:
Latest model milestone
[P-TET-000019-3.zip](https://github.com/tetrabiodistributed/project-tetra/files/5086555/P-TET-000019-3.zip)