tetrabiodistributed/project-tetra #99: P-TET-000022 Design Fixes

mechatroniks-git, 2020-07-21:
- [ ] Check o-ring groove
- [ ] OD of bore as it mated with P-TET-000023 was too tight, need to reduce to avoid interference
- [ ] Reduce amount of material used in part, cost saving

cyunis, 2020-07-23:
Videos of sanding part to fit (and cleaning after) on 7/20: 
[Archive.zip](https://github.com/tetrabiodistributed/project-tetra/files/4963879/Archive.zip)

mmroden, 2020-07-27:
2 day estimate + 2 day revision

jgrano, 2020-07-30:
https://github.com/tetrabiodistributed/project-tetra/blob/master/02_Development/Prototype_release/Alpha/Parts_to_print/P-TET-000022-1.stl

Burhan-Q, 2020-07-31:
@DHHwangPhD please comment when printing has started

Burhan-Q, 2020-08-01:
As of today. @DHHwangPhD printed part and is curing. Will deliver to @mechatroniks-git as soon as possible

mechatroniks-git, 2020-08-06:
P-TET-000022 is now replaced by A-TET-000018, ready for review
[image]

mechatroniks-git, 2020-08-06:
I would advise holding off on printing the part until we can review print from 5 days ago

Burhan-Q, 2020-08-06:
> I would advise holding off on printing the part until we can review print from 5 days ago

We can hold off, but I recall @DHHwangPhD saying that the last print didn't go well (which is why this change was made by @jgrano). I'll double check with him to see how he wants to proceed with this part.

Burhan-Q, 2020-08-06:
@mechatroniks-git I spoke with @DHHwangPhD and he said that the previous model did not work and would like to print this version. I did a quick review and thing appear okay. I'm going to share the new cap design with him as well as the Tetra valve body #100 part for testing. 

Per FormLabs post about threaded inserts for resin prints, they say that the inserts should go in before curing. Of course we could also drill out the cap and glue/epoxy it in after curing, but perhaps it would be worth having Darryl get some from you or order some for the other prints.

Burhan-Q, 2020-08-06:
@DHHwangPhD ready for print testing. Please comment when you've started printing.

Plunger
[P-TET-000030-0.zip](https://github.com/tetrabiodistributed/project-tetra/files/5037913/P-TET-000030-0.zip)

Cap
[P-TET-000031-0.zip](https://github.com/tetrabiodistributed/project-tetra/files/5037914/P-TET-000031-0.zip)