tetrabiodistributed/project-tetra #108: P-TET-000001/2/3 Piping manifold design updates

Burhan-Q, 2020-07-27:
Branch of issue #106 

Need to determine Glue for tubing parts

Darryl & Burhan (who will check with FormLabs)

- possible to use liquid resin and cure with UV flashlight
- potential to use E6000 as a glue

Burhan-Q, 2020-07-30:
[image]
From [FormLabs blog article](https://formlabs.com/blog/how-to-create-models-larger-than-your-3d-printers-build-volume/?utm_source=pinshape&utm_medium=blog&utm_campaign=adhesives)

Burhan-Q, 2020-07-30:
[adhesive strengths for 3D printed bars](https://pinshape.com/blog/discover-the-best-adhesive-for-your-project/)
[FormLabs forum comment](https://forum.formlabs.com/t/is-there-an-instant-glue-for-formlab-resin-prints/8978/3)
[FL forum discussion on laser resin gluing](https://forum.formlabs.com/t/deeper-laser-gluing-curing-discussion/21792)
[FL general gluing discussion](https://forum.formlabs.com/t/how-do-you-glue-pieces-together/6168/10)
[FL more gluing discussions](https://forum.formlabs.com/t/glue-for-assembly/1856)

Burhan-Q, 2020-07-31:
@DHHwangPhD have any input on which direction we should go? I'm thinking a normal CA super glue might make the most sense for now.

Burhan-Q, 2020-08-01:
Spoke with @DHHwangPhD. Determined 2 means for adhering resin parts to keep air tight:

Primary: Attach with printer resin and cure with handheld UV light
Secondary: Attach using 5 minute epoxy

Darryl has UV light on order, but not likely to arrive in time for Alpha prototype assembly. @mechatroniks-git do you have or could you obtain 5 minute epoxy for assembly?

mechatroniks-git, 2020-08-03:
@Burhan-Q I have a few 2-part epoxy options

Burhan-Q, 2020-08-03:
Excellent @mechatroniks-git, thank you. Do you think we're ready to close this issue?

closed 2020-08-03 by mechatroniks-git