tetrabiodistributed/project-tetra #110: [DRAFT] Sprint-2 Alpha parts printing queue

Burhan-Q, 2020-07-29:
# DRAFT # 
This Issue is a draft outline for a means to track our printing queue for the Alpha Prototype Sprint-2. Once approved for use, this ticket will be closed and new issue will be created to actually use for production. Please comment on this Issue for discussion of changes or suggestions needed, or to communicate your approval for using this format.

----------------------------------------------------------------------------------------------------------------------------------------------

**Files ready for release will be listed in the first section below and contain an active link to the file for printing ONLY when that part has been released.**

## Instructions ##
1. Find part ready for printing in list below
2. If you're able to print a part, click and drag the checkbox for that part under your name.
   - Move under Printing in queue if you can print the part but can't begin immediately
   - Move to Printing in progress if print can start immediately. 
3. After all quantities of part have been printed, click checkbox to mark as complete
4. Add comment to this issue with the following information for **each** Part:
   - Type, Brand, and Name of Material used
   - Layer height, InFill %, Critical Printer settings
   - Any important notes or photos with respect to printing the part

### Released Parts for printing ###
- [x] _reserve_
- [ ] \[SLA/FDM P-TET-000000-X Description xN](STL_file_link)
- [ ] \[SLA/FDM P-TET-000000-X Description xN](STL_file_link)
- [ ] \[SLA/FDM P-TET-000000-X Description xN](STL_file_link)

   ## Darryl SLA & FDM ##
### Printing in progress ###
- [x] _reserve_

### Printing in queue ###
- [x] _reserve_

   ## Mark FDM ##
### Printing in progress ###
- [x] _reserve_
 
### Printing in queue ###
- [x] _reserve_

   ## Patrick FDM ##
### Printing in progress ###
- [x] _reserve_

### Printing in queue ###
- [x] _reserve_

   ## OTHER SLA/FDM ##
### Printing in progress ###
- [x] _reserve_

### Printing in queue ###
- [x] _reserve_

Burhan-Q, 2020-07-31:
@jgrano @mechatroniks-git @DHHwangPhD

Please check this out and comment with your thoughts about using this format for tracking the print queue

Burhan-Q, 2020-07-31:
@chrismolinari456

chrismolinari456, 2020-07-31:
I'm a bit confused; will this be a separate project board that will have all of our names and the parts? Or does the issue itself support a drag/ drop interface?
I like the shift personally; it keeps everything in one place and cards are easier to edit/ move than cells of text

Burhan-Q, 2020-07-31:
> I'm a bit confused; will this be a separate project board that will have all of our names and the parts? Or does the issue itself support a drag/ drop interface?

Yes it does, if you hover over one of the checkboxes, you should see 3 horizontal bars pop up on the left, you can click and drag those (if you do, please move it back so others see the original)

chrismolinari456, 2020-07-31:
Just checked it out on a test repo and it seems to work pretty well, but for some reason I can't edit it here. Maybe because I haven't done contributed to the main repo

DHHwangPhD, 2020-07-31:
This works for me.

Burhan-Q, 2020-08-01:
> Just checked it out on a test repo and it seems to work pretty well, but for some reason I can't edit it here. Maybe because I haven't done contributed to the main repo

Hm, maybe double check with @capsulecorplab on permissions?

Burhan-Q, 2020-08-01:
Moving to Blocked. Still waiting on feedback from @mechatroniks-git @mmroden and @jgrano; looking for Go/No-Go to use instead of printer spreadsheet and/or recommendations if lacking any features

Burhan-Q, 2020-08-05:
As per discussion from meeting August 5, 2020; we will continue using the [Google Sheets document](https://docs.google.com/spreadsheets/d/1FWTYRxglWDqIK_xI6HlBGKUegICQRsXe5NHCJTlTlHs/edit#gid=180162121&range=A3) for print tracking. 
@capsulecorplab or @mmroden should this Issue be closed? Or should I Label this as "Stale"?

capsulecorplab, 2020-08-06:
I would close it, commenting it as stale or archived

Burhan-Q, 2020-08-06:
stale or archived

closed 2020-08-06 by Burhan-Q