tetrabiodistributed/project-tetra #112: A-TET-001004 Tetra PCB Assembly 5V Regulator Issue

mechatroniks-git, 2020-08-03:
Board serial # 00 is not outputting 5VDC

mechatroniks-git, 2020-08-03:
https://github.com/tetrabiodistributed/project-tetra/tree/master/02_Development/FlowMeter/RPi%20Board

Datasheet: http://www.xlsemi.com/datasheet/XL4015%20datasheet.pdf

hydronics2, 2020-08-03:
@mechatroniks-git I just reviewed the pinout and reference schematic. The pinout is correct. All the parts seem placed as shown. I am missing this cap on the FB line. (33nf)
[image]

[image]

hydronics2, 2020-08-03:
I figured it out I think... they installed the diode in backwards... The orientation was correct when I reviewed it online but something must have happened at the boardshop...!  I'll get it swapped around and see what happens... 
[image]

hydronics2, 2020-08-03:
Ok... we got 5V! woohoo...  I turned the diode around and added a 33nF cap along side the 10K resistor (not sure if the cap is 100% necessary but following the data sheet)
[image]

notice I made the jumper... so now you have 5V LED lit and the RPI 5V LED lit.... + the 12V LED lit...so that's 3 indicator LEDs (all accept the 3V rail which comes from the PI).   Pulling 17ma at 12V. 

[image]

mechatroniks-git, 2020-08-04:
Spun the diode and added a 10nF cap (didn't have 33nF), getting 5V

[image]

closed 2020-08-04 by mechatroniks-git