tetrabiodistributed/project-tetra #113: Sensirion control PCB adjustments

Burhan-Q, 2020-08-03:
Phoenix connector needs to move inboard to avoid issues with mounting to Sensirion sensor.

### Current distance (center to center):
Phoenix pins to pogo pins = 9.705mm
Phoenix pins to board face = 6.721mm

### Constraints:
Distance pogo pins to front of sensirion cut out = 9.597mm - 0.862mm = 8.735mm [bumpouts on inside of Sensirion cutout]
Phoenix connector to control board PCB component = 1.938mm (max distance phoenix connector can move backwards, see image below)

#### Image of interference
[image]

#### Pogo pins to front wall (subtract 0.862mm for bump-out, center to edge)
[image]

#### Current space between pogo pins and phoenix connector pins (center to center)
[image]

#### Current phoenix pins to board (center to face):
[image]

#### Phoenix pins to Sensirion PCB component:
[image]

#### Second view:
[image]

#### Space between rear of (current) phoenix connector to control board PCB component:
[image]

Burhan-Q, 2020-08-03:
Uploaded these to Slack, but also posting them here
[STEP_files_requested.zip](https://github.com/tetrabiodistributed/project-tetra/files/5018030/STEP_files_requested.zip)

Burhan-Q, 2020-08-03:
### Pogo pins and phoenix connector pins
Center to center distance on @mechatroniks-git board model = 8.43mm
[image]

### Phoenix connector pins to board face
Center to face distance = 8.00mm
[image]

hydronics2, 2020-08-04:
@Burhan-Q ... created a pull request with the new files. thanks for the pics!... and that fusion 360 model is worth a 1000 words.

closed 2020-08-05 by fpdotmonkey