tetrabiodistributed/project-tetra #115: P-TET-000020 Tetra Shutoff Valve Cap needs clearance

mechatroniks-git, 2020-08-04:
Valve cap needs following tweaks
- [ ] Clearance to fit onto valve body P-TET-000023 & P-TET-000018
- [ ] Clearance to fit over the P-TET-000019 without binding
- [ ] Clearance to allow P-TET-000021 to freely rotate

Some of this may be due to printing imperfections, but the design should easily accommodate imperfect printing while allowing functionality. 

[image]

It appears that the current Fusion model has clearance, but the -0 part I printed does not. Need to make sure what is in Fusion is not the same as the -0, then add the necessary clearance (3rd item on lis) and release part for printing. 

[image]

jgrano, 2020-08-04:
@mechatroniks-git did you pull the part from the spread sheet or the main model? I uploaded the parts, printed on high resolution on a taz5 with abs and it went together smoothly for me.

Burhan-Q, 2020-08-04:
@mechatroniks-git something weird is happening the assembly model when you open it. The handle is positioned as such when first opened (like in your image above)
[image]

But this position is incorrect, and if you move the handle with a click and drag, you'll notice the cap flips to the correct orientation.
[image]

No more overlapping parts when viewing cross section
[image]

mechatroniks-git, 2020-08-04:
I've only printed the -0 part. 

The latest knob has a max dia of 21.57 - 21.54, depending on the measuring spot. The design is 21.50, so I'm a bit oversized, but well within tolerance. 

[image]

The  mating part is also at 21.50, and should be 21.7. I can adjust the design and print.

[image]

mechatroniks-git, 2020-08-04:
Printing this one now
[image]

Burhan-Q, 2020-08-04:
@mechatroniks-git should this issue get updated to Testing?

mechatroniks-git, 2020-08-04:
The -2 part feels good to me. Little bit of vertical play in the valve assembly, maybe .3mm, but probably acceptable. 

I think testing is done, ready for review @Burhan-Q 

[image]

Burhan-Q, 2020-08-04:
> Valve cap needs following tweaks
> 
> * [ ]  Clearance to fit onto valve body P-TET-000023 & P-TET-000018
> * [x]  Clearance to fit over the P-TET-000019 without binding
> * [x]  Clearance to allow P-TET-000021 to freely rotate

@mechatroniks-git looks like you accomplished the two items above (based on your most recent comment). Not sure if there was a typo on the first bullet point P-TET-000023 is the Tetra valve and is not related to the shutoff valve cap, assuming you meant the shutoff valve body and knob P-TET000018 and P-TET-000021 respectively. 

All else looks good, expect the filename needs to be updated to match the revision number -02. Once complete, ready for approval @jgrano.

jgrano, 2020-08-04:
Looks awesome guys

Burhan-Q, 2020-08-05:
> Looks awesome guys

Closing ticket

Burhan-Q, 2020-08-05:
Reopened since file needs to be added to repo. Moved "Closed" bucket on Project board since work is complete otherwise

mmroden, 2020-08-05:
if the file isn't in the repo, then the ticket is not closable.

commit b06d9ab7 (2020-08-31): Design files Alpha sprint 2

closes #64
closes #67
closes #96
closes #95
closes #115
PR #128 (2020-09-01): Alpha sprint p2
closes #64 
closes #67 
closes #96 
closes #115

Accidentally duplicated previous PR inclusion of drawing template files, no changes should have been made
commit de3b20d1 (2020-09-01): Alpha sprint p2 (#128)

* Create folder

* Issue #98

* Issue #97

* Clean up release vs testing files

Add Archive folder for previous revision models
Add Testing folder for models not released but ready to test
Migrated For_Printing folder with currently released models

P-TET-000019-2 Issue #98
P-TET-000021-1 Issue #97
P-TET-000022-1 Issue #99
P-TET-000023-1 Issue #100
P-TET-000028-0 New iteration of Tetra valve with steps added to outlet

* Updated P-TET-000030 to -1

* Design files Alpha sprint 2

closes #64
closes #67
closes #96
closes #95
closes #115
closed 2020-09-01 by Burhan-Q