tetrabiodistributed/project-tetra #118: A-TET-001014 Expiratory Support mount Design Change

Burhan-Q, 2020-08-05:
Can we make these parts the same? Also, the diameter of each of these is different?

[image]

_Originally posted by @mechatroniks-git in https://github.com/tetrabiodistributed/project-tetra/issues/87#issuecomment-669414512_

Judoguy12, 2020-08-06:
Done @mechatroniks-git

mmroden, 2021-02-08:
Needs a reviewer

mechatroniks-git, 2021-02-10:
Not complete. The diameter was adjusted, but the parts are still unique part #s (P-TET-000011 & P-TET-000013). They should be the same. We should add embossed part numbers to the model as well. To make work, we need to modify P-TET-000014 to match the width and screw pattern.

[image]

Judoguy12, 2021-02-13:
Should be all done @mechatroniks-git