tetrabiodistributed/project-tetra #119: Verify new pressure sensor assembly fits inside housing A-TET-001017

Burhan-Q, 2020-08-07:
New pressure sensor communication PCB #113 & #114 and wiring harness #67 needs to be checked for clearances in full assembly model.

- [ ] import new pressure sensor assembly A-TET-001017 to Alpha prototype assembly
- [ ] join A-TET-001017 with piping in correct location(s)
- [ ] verify that A-TET-001017 doesn't interfere with other components or creates other assembly or usability issues

Assignments:
- Work assigned to:
     @Burhan-Q 
- Reviewer (select name):
     - [ ] @mechatroniks-git 
     - [ ] @Judoguy12 
     - [ ] @jgrano 
- Approver (select name):
     - [ ] @mechatroniks-git 
     - [ ] @Judoguy12 
     - [ ] @jgrano

Burhan-Q, 2020-08-11:
Clearance to housing enclosure is 2.637 mm. No other interference observed. Assembly may be rotated to assist with other potential collisions (with any parts currently not added to model).

[image]

Judoguy12, 2020-08-17:
Reviewed 17/08/2020 All changes have been made ready for approval

mmroden, 2021-02-08:
Housing is now "nice to have"

mmroden, 2021-02-08:
PCB grew 1 mm, this issue is still open, and not nice-to-have