tetrabiodistributed/project-tetra #121: P-TET-000025 Inspiratory bias

Burhan-Q, 2020-08-15:
- [ ] cleanup designs, remodel if needed
- [ ] add part number emboss
- [ ] share STLs for printing
- [ ] print test and review
    @jgrano will print these

Burhan-Q, 2020-08-15:
Ready to print
[P-TET-000025-1.zip](https://github.com/tetrabiodistributed/project-tetra/files/5079369/P-TET-000025-1.zip)

Burhan-Q, 2020-08-15:
Work completed 2020-08-15; reassign to @jgrano for printing

Burhan-Q, 2020-08-24:
### Update ###
@Burhan-Q made minor adjustments to prior design, @jgrano finalized designs; these were sent to @DHHwangPhD for printing

### Notes ###
- will need one-way valve flapper installed
- Grooves (x2) for O-Ring have outside diameter of 28.50 mm and are 1.05 mm deep

**Inspiratory bias circuit**
[image]

mmroden, 2021-02-08:
Jamie to test