tetrabiodistributed/project-tetra #122: P-TET-000026 Expiratory bias

Burhan-Q, 2020-08-15:
- [ ] cleanup designs, remodel if needed
- [ ] add part number emboss
- [ ] share STLs for printing
- [ ] print test and review
     @jgrano will print these

Burhan-Q, 2020-08-15:
Ready to print
[P-TET-000026-1.zip](https://github.com/tetrabiodistributed/project-tetra/files/5079370/P-TET-000026-1.zip)

Burhan-Q, 2020-08-15:
Work completed 2020-08-14; reassign to @jgrano for printing

Burhan-Q, 2020-08-24:
### Update ###
@Burhan-Q made minor adjustments to prior design, @jgrano finalized designs; these were sent to @DHHwangPhD for printing

### Notes ###
- P-TET-000026 is now multibody (3 components), need to determine if these should have separate part numbers

**1 of 3 components highlighted**
[image]

mmroden, 2021-02-08:
fitted, not tested