tetrabiodistributed/project-tetra #123: Create P-TET-000032 Rear port cover

Burhan-Q, 2020-08-18:
Relies on completion of #121 and #122 (bias components).

Specifications:
- [ ] conforms to ISO-5356 fitting standard
- [ ] attaches to rear opening on housing (?)

Burhan-Q, 2020-08-24:
### Update ###
@jgrano created parts and added to assembly, these were sent to @DHHwangPhD for printing with use in the Alpha prototype

### Notes ###
- ISO port adapters for rear panel are using P-TET-000032, need to change descriptor and create new part number (PN) for rear cover; will create new Issue for the new component and document notes there
- O-Ring groove is 1.05 mm deep with seat diameter of 28.50 mm
- Taper angle as measured in image below is 1.432 degrees (agrees with ISO spec taper ratio 1/40)

### Review ###
- [ ] 1. Need to add visible embossing of PN on components
- [ ] 2. Taper distance until outside diameter (OD) is 22.314 mm; 13.4mm (see photo); ISO spec is for 22.37 mm at distance of 15 mm
- [ ] 3. O-Ring spec is required (add comment to thread); is it possible to modify design to match an existing O-Ring spec?
- [ ] 4. Add locking coupler to prevent untwisting of port connector (not needed if bonded chemically)?

**OD taper angle distance**
[image]

**O-ring groove**
[image]

**Rear ports w/ cover**
[image]

mmroden, 2021-02-08:
being superceded by nylon, Jamie can test