tetrabiodistributed/project-tetra #124: P-TET-000033 Rear port cover

Burhan-Q, 2020-08-24:
Need to create Rear port cover part.

It will be a housing cover plate for rear ports adapters that connect to ventilator #123

Burhan-Q, 2020-08-24:
This ticket was opened after the work by @jgrano was completed. Creating ticket separate from the Rear port adapters #123 which was initially established for creation of rear cover

Burhan-Q, 2020-08-24:
### Notes ###
- Openings cut at angle to match housing rear panel
- No apparent mechanism designed for attaching plate to housing, options: bond with epoxy, glue, or drill and screw
- Clearance between ports and opening ~2 mm
- No part number embossed on body
- Status (per week of August 24, 2020)
    - [ ] Ready for review
    - [ ] Design requires more updates

**Rear port cover**
[image]

**Port clearance**
[image]

Burhan-Q, 2020-08-26:
@mechatroniks-git made some modifications to the design and iterated to P-TET-000033-1 (see image). He will also be printing this for the Alpha build, so the Assignment for this issue has been updated and the [G-Sheet](https://docs.google.com/spreadsheets/d/1FWTYRxglWDqIK_xI6HlBGKUegICQRsXe5NHCJTlTlHs/edit#gid=563995584&range=A1) will also reflect this. These details come from a Slack DM at ~1700-1715 EDT on August 26, 2020.

[image]

mmroden, 2021-02-08:
Good for now, will reopen if we need it

closed 2021-02-08 by mmroden