tetrabiodistributed/project-tetra #133: Alpha Prototype rework

Burhan-Q, 2020-10-05:
@jgrano taking on all tasks pertaining to Prototype design updates/changes and final assembly.

## Goals ##
- Restructure Fusion files based on PDM discussion
- Find all fitment issues, update designs as needed
- Modify internal components and/or enclosure design as needed for optimizing assembly
- Optimize any designs for manufacturing or assembly
- Manufacture and fully assemble prototype
- Verify mechanical components functionality and perform basic leak check for all airway components

## Expected outcomes ##
- Documentation of all issues discovered and modifications made (images, notes, etc.)
- Fully assembled prototype passing basic tests for mechanical functionality and airtightness (minimal to no leaking)
- Completed prototype assembly shipped to California for electronics and functionality testing

Burhan-Q, 2020-10-05:
Identified overlapping components (inside of CAD)

## Sensirion to Shutoff valve overlap ##
[image]

## Sensirion to ISO adapter overlap ##
[image]

## Tetra to Shutoff valve overlaps ##
[image]

Judoguy12, 2020-10-05:
Following