tetrabiodistributed/project-tetra #140: Write Tetra's Testing specification

Judoguy12, 2020-11-18:
Write down the tests we want tetra to undergo and define pass criteria

Judoguy12, 2020-11-18:
Will be transferred to the website

https://docs.google.com/document/d/1woSVfRFWcU1IHEs3X66P7dTMBY_45qS-Xk9WlSRjt7Q/edit

Burhan-Q, 2020-11-21:
Also see the following documents for reference on ventilator and vent-sharing testing and/or protocols:

[Ventilator Service manual](https://www.medonegroup.com/pdf/manuals/techManuals/Maquet-Servo-i-Service-Manual.pdf) - see Troubleshooting section, other sections good reference as well
[FDA Criteria for safety, performance, and labeling](https://www.fda.gov/media/136437/download) - includes some info about vent splitting on pg4
[NY Hospital Vent-sharing protocol](https://www.gnyha.org/wp-content/uploads/2020/03/Ventilator-Sharing-Protocol-Dual-Patient-Ventilation-with-a-Single-Mechanical-Ventilator-for-Use-during-Critical-Ventilator-Shortages.pdf) - includes some testing info

closed 2021-05-15 by mmroden