tetrabiodistributed/project-tetra #146: Pi HAT PCB review

darraghbr, 2020-12-11:
Upon reviewing the Pi HAT for the tetra project I have come up with a couple of potential issues.

1 - The pullup resistors between the pressure sensors and the I2C multiplexer seem too high a value, according to the application note and some calcs I would expect them to be in the range of 1 - 5 KOhm depending upon fast mode being enabled. 

2 - The I2C address selection pins on the I2C multiplexer are currently connected to GND or 3.3V via a 10K resistor, the application note suggests that they are connected directly to 3.3V or GND, this is probably not critical. 

3 - The board layout has some potential clearance issues between the edge of the board and the mounting holes, this issue may or may not be relevant depending on the board house. 

4 - The PCB also has some curious wire stubs that don’t appear to be intentional, they should be replaced with polygon, this is also not critical. 

5 - The application note for the ADC suggests crossing the analogue and digital traces at 90° to one another, and to keep the traces as short as possible, the layout at the moment doesn’t seem ideal for these. If we see much poorer accuracy on ADC channel 1 and 2 compared to 3 and 4 this will probably be why. If there is a suspicion are close to the tolerance on these sensors it may warrant more investigation.

mmroden, 2020-12-11:
Please make each one of these into separate issues so we can be sure that
they all get addressed; when there are many issues in the same ticket,
chances are good that one or two will get lost in the shuffle.

closed 2020-12-11 by darraghbr

mmroden, 2020-12-11:
Thank you for the split!  I don't want to lose anything you're saying