tetrabiodistributed/project-tetra #147: Pi HAT PCB multiplexer resistors incorrect value

darraghbr, 2020-12-11:
The pullup resistors between the pressure sensors and the I2C multiplexer seem too high a value, according to the application note and some calcs I would expect them to be in the range of 1 - 5 KOhm depending upon fast mode being enabled.

mechatroniks-git, 2020-12-12:
@hydronics2 or @fpdotmonkey Can you tell us if fast mode is being used?

darraghbr, 2020-12-12:
Even with fast mode enabled and nearly the highest bus capacitance that the chip can tolerate the range for the appropriate resistor is between 900 and ~ 1012 Ohms. I would just replace them with a 1K resistor.

mechatroniks-git, 2020-12-12:
I didn't run the formula, but page 23 of the [datasheet](https://www.ti.com/lit/ds/symlink/tca9548a.pdf?ts=1607793644356&ref_url=https%253A%252F%252Fwww.google.com%252F) shows this:
[image]
I agree with your recommendation to reduce the 10K resistors. 1K would cover the range, if we are using fast mode. Downside is slightly higher drive currents, and possible higher EMI (di/dt). If we are never running fast mode, then we should stick to a higher R in the 3K range, to play it conservative, for the sake of EMI.

hydronics2, 2020-12-13:
wow... that's a great graph. Thanks for the feedback. I don't know the i2c
speed but we should be pushing for slow speeds as much as possible for the
off-board sensiron sensors.

I typically push using 10K for everything to simplify the board design.
Weak pullups  (10k) should be fine for the onboard SPL006 pressure
sensors but the sensirion sensors (off board) probably makes sense to drop
these pullups to something less than 10K.

I'll drop to 3K per your suggestion for all i2c just to keep things similar.

darraghbr, 2020-12-14:
Issue resolved by #152

closed 2020-12-19 by darraghbr