tetrabiodistributed/project-tetra #148: I2C address selection pins not connected in accordance with application note

darraghbr, 2020-12-11:
The I2C address selection pins on the I2C multiplexer are currently connected to GND or 3.3V via a 10K resistor, the application note suggests that they are connected directly to 3.3V or GND, this is probably not critical.

mechatroniks-git, 2020-12-12:
On page 5&6 of [datasheet](https://www.ti.com/lit/ds/symlink/tca9548a.pdf?ts=1607713427169&ref_url=https%253A%252F%252Fwww.ti.com%252Fstore%252Fti%252Fen%252Fp%252Fproduct%252F%253Fp%253DTCA9548APWR)

[image]

[image]

shows 1uA of current on these pins, having a 10 to gnd or +Vcc will not affect the logic level. The 10k can be left in place, as this is usually considered good practice to tie logic pins to power rails in this manner.

closed 2020-12-12 by mechatroniks-git

darraghbr, 2020-12-12:
While I don't disagree with you that having a 10K resistor won't hurt, the data sheet on page 4 states this. That was the basis for me making this issue.

[image]

mechatroniks-git, 2020-12-12:
The "directly" comment here is certainly confusing. 

I found this:  [eval board](https://www.ti.com/lit/ug/slvua73/slvua73.pdf?ts=1607799198136&ref_url=https%253A%252F%252Fwww.ti.com%252Ftool%252FTCA9548AEVM), Here TI is using 10K to gnd/vcc.
[image]