tetrabiodistributed/project-tetra #149: Board mounting holes too close to the edge of the PCB

darraghbr, 2020-12-11:
The board layout has some potential clearance issues between the edge of the board and the mounting holes, this issue may or may not be relevant depending on the board house.

mechatroniks-git, 2020-12-12:
@darraghbr Can you share your eagle DRU file? We should be evaluating the design rules with a common file.

darraghbr, 2020-12-12:
I am using the dru file presented in the JLCPCB official GitHub repo, which can be found here:

https://github.com/JLCPCBofficial/jlcpcb-eagle/tree/master/design%20rules

It turns out that I misread the clearances slightly and it is actually an issue between a wire and a hole rather than a hole and something on the dimension layer. I will edit the title and initial comment to reflect this.

mechatroniks-git, 2020-12-12:
Ah ok, I had downloaded someone's JLCPCB from github, but upon inspection didn't match their [capabilities](https://jlcpcb.com/capabilities/Capabilities) page. Even the official one is not consistent, for example, website offers 0.2mm minimum drill, DRU is at 0.3mm. 

In any case, using their official DRU, looking at the file:

- I see the 3 clearance errors on the bottom copper, which I agree they should be fixed. 
- The 6 overlap errors with the ground vias, only way to fix those is to make the vias part of the library package, but it's not a true error and could be ignored without issue. 
> I don't have the th_libraries.lbr file, so @hydronics2 would have to provide for me to fix

darraghbr, 2020-12-13:
Yes I recognised the thermal vias on the power supply were just a small issue with the library. That's definitely in the non critical category.

hydronics2, 2020-12-13:
- yea it looks like the JLCP allows signal to edge of 0.2mm or 8mil. I
don't like stretching my polygon ground planes to the max allowable extent
so I have it set at 15mil in the DRU. Setting the DRU to 15mil I do see
some errors for 5 plated holes on the right side of the board that go away
when I set the DRU to 8mil. Can you send me a screen shot of the "3
clearance errors on the bottom side"?
[image: image.png]
[image: image.png]

- 6 overlap errors... I researched a bit adding vias to the part library.
From my autodesk googling it seemed perfectly allowable to add vias
manually in the Schematic/Board editor.  To get get rid of the DRU error
you can change the clearance between same signals as follows:
[image: image.png]

hydronics2, 2020-12-13:
do these images show up?
[image]
[image]
[image]

darraghbr, 2020-12-14:
Issue resolved by #152

closed 2020-12-19 by darraghbr