tetrabiodistributed/project-tetra #150: Potential issue with noise on the ADC

darraghbr, 2020-12-11:
The application note for the ADC suggests crossing the analogue and digital traces at 90° to one another, and to keep the traces as short as possible, the layout at the moment doesn’t seem ideal for these. If we see much poorer accuracy on ADC channel 1 and 2 compared to 3 and 4 this will probably be why. If there is a suspicion that we are close to the tolerance on these sensors it may warrant more investigation.

hydronics2, 2020-12-13:
Yea I agree the routing on the ADC could be better although I've not studied how the digital/analog lines cross each other.  I think we should keep this issue open and evaluate it once we get feedback on the sensors. The layout was for a quick prototype and right now there there seems little interest in moving forward with the ADC and associated automotive sensor so I'm hesitant to spend any time on this.

mmroden, 2021-02-08:
Does not need to be done for a first deployment, not sure if the ADCs will be used