tetrabiodistributed/project-tetra #151: Wire stubs found in PCB design

darraghbr, 2020-12-11:
Upon doing a DRC check the PCB has some curious wire stubs that don’t appear to be intentional, they should be replaced with polygon, this is probably not critical.

mechatroniks-git, 2020-12-12:
I see 6 wire stub warnings, some in the GND plane get covered by GND polygon. None are an issue, but certainly worth cleaning up when fixing the other errors.

hydronics2, 2020-12-13:
I tried deleting the wire stubs and I get air-wire warnings (un-routed
traces). This happens to me from time to time when I'm routing and
rerouting work between revisions... finicky Ealge thing... I'm not sure how
to get rid of the wire-stubs and because air-wire warnings are worse than
wire-stub warnings I'm inclined to leave them. If anyone can get rid of the
wire stubs, please go ahead.

darraghbr, 2020-12-14:
Issue resolved by #152

closed 2020-12-19 by darraghbr