tetrabiodistributed/project-tetra #153: P-TET-000026 Dimension problems

Judoguy12, 2020-12-18:
This part has some wrong dimensions, the 3 way tube has a different dimension for each output, these need to be standardised. Also it has an orange groove but the orang is not specified

Burhan-Q, 2020-12-22:
@Judoguy12 can you provide details on which dimensions are incorrect/different? Looking at the model (in the Alpha 2.0 folder) looks like they're all the same diameters. 

Also, the ID for the o-ring groove looks to be 26.40 mm, so the o-ring to use is likely V-MCM-9262K527 - Buna-N O-Ring 1.5 mm Wide, 26 mm ID

Judoguy12, 2020-12-22:
Yeah the right angle pipe that connects to the bias peice was much harder
to fit that the right angles on the edges of the inspiratory and expiratory
circuits even though they look to be largely the same part.

Thanks for the oring size

mmroden, 2021-02-08:
Jamie will check it