tetrabiodistributed/project-tetra #159: Missing item on Pi HAT BOM

darraghbr, 2021-01-31:
We are missing the 2x20 header strip that goes between the Pi and the HAT. 

I think this is roughly the correct part number.

https://www.mouser.co.uk/ProductDetail/TE-Connectivity/2-534998-0/?qs=%2Fha2pyFaduiXhbXo1Q7kVEqN1S1BJj7DwXrciKekKYCi1xy6LVwXqQ==&utm_source=octopart&utm_medium=aggregator&utm_campaign=571-25349980&utm_content=TE+Connectivity

darraghbr, 2021-01-31:
@mechatroniks-git

hydronics2, 2021-01-31:
That looks right to me. This is the one I used: https://lcsc.com/product-detail/Pin-Header-Female-Header_BOOMELE-Boom-Precision-Elec-C50982_C50982.html

mechatroniks-git, 2021-02-01:
The TE part will work just fine. 
[image]

closed 2021-02-03 by Judoguy12