tetrabiodistributed/project-tetra #163: [Hardware] Review drawings for push fit manifold and tube adaptors before release

darraghbr, 2021-02-23:
I have generated a version of the pressure sensing manifold and 22 mm tube adaptor in order to be compatible with the push fit nylon version of the Tetra. It needs reviewing before it is issue for manufacture.

Please see attached pdfs of the components. A dxf file will be generated and issued along with the drawings.

[D-TET-001105-0 Push Fit Manifold Drawing.pdf](https://github.com/tetrabiodistributed/project-tetra/files/6026938/D-TET-001105-0.Push.Fit.Manifold.Drawing.pdf)

[D-TET-001106-0 Push fit to 22mm hose adaptor Drawing.pdf](https://github.com/tetrabiodistributed/project-tetra/files/6026939/D-TET-001106-0.Push.fit.to.22mm.hose.adaptor.Drawing.pdf)

darraghbr, 2021-02-23:
@mmroden who should this get assigned to?

mechatroniks-git, 2021-02-24:
1. The assembly drawing should not have part dimensions, this should only be on the part drawing. Here we can have dims on the assembly, between parts, or overall dims are ok too. 
2.  Part number not meeting format, should be M-LEGRIS-XXX & M-???-BS111
[image]

Specific to the assembly drawing - assembly should not have a material property, as it consists of parts being assembled. 

Specific to the parts (which should be on the part drawings):
3. 6061-T6 is more commonly available, this may be harder to source and more expensive?
4. With CNC, need to put in a inside corner radius to that's slightly larger than the tool radius. So if using a 4mm dia tool, set the radius to 2.5mm. (more of a part note, not assembly)
[image]

mechatroniks-git, 2021-02-24:
D-TET-001106 same issues with part number missing M-LEGRIS. Also, need part drawing for the aluminum part. This looks like a CNC lathe part, so need corner radius 
[image]

Does the Legris o-ring need any o-ring grease during assembly? If so, make a note.

darraghbr, 2021-02-24:
The ID on the 001106 part I would suspect will be bored out, so radius on the end will be minimal. I have added a comment on the OD of the part to suggest the feature is non-critical. 

On 001105 6061-T6 is not a material we can source readily, 6082 is the alloy we can get hold of here. I have also added a note about the external profile being non-critical. With an ISO standard drawing it's pretty normal to have assembly and parts mixed together with dimensions, and ultimately with .step files being issued as well it should not matter. 

Please find attached new versions of the drawings.

[D-TET-001106-0 Push fit to 22mm hose adaptor Drawing.pdf](https://github.com/tetrabiodistributed/project-tetra/files/6038054/D-TET-001106-0.Push.fit.to.22mm.hose.adaptor.Drawing.pdf)
[D-TET-001105-0 Push Fit Manifold Drawing.pdf](https://github.com/tetrabiodistributed/project-tetra/files/6038055/D-TET-001105-0.Push.Fit.Manifold.Drawing.pdf)

Judoguy12, 2021-02-26:
@mechatroniks-git Are these drawings up to snuff?

mechatroniks-git, 2021-02-26:
I still say that an assembly drawing shouldn't have any part dimensions, other than what's needed for assembly. We need to delete those and create a part drawing with those dimensions, so we have a single source of truth.

darraghbr, 2021-03-02:
Ultimately a drawing is there to convey information, and unless a drawing is difficult or confusing to read what we are talking about is a style choice. 

Removing information from a drawing and generating twice as many drawings as is necessary does not solve any problems that I can see and in my opinion dilutes the concept of a single source of truth.

This is my argument to keep them how they are, @mmroden please make a call and I will follow it.

mmroden, 2021-03-02:
I'm not sure I understand completely what I'm being asked to adjudicate here.  It sounds like there's a question of having an assembly drawing and a part drawing?  Should we discuss in the meeting tomorrow?  What's the difficulty of having a part drawing with materials distinct from an assembly drawing with no materials but just how parts are expected to fit together?  Don't they convey different information?

I also think we should be putting the drawings up in PRs rather than issues, so that we have them in the repo immediately after approval, rather than a second PR approval process.

mechatroniks-git, 2021-03-03:
I've never sent assembly drawings to a machine shop, but I've never used shops in Europe, perhaps it's common there?  Here, they want part drawings if they are doing to make parts. But if it works, leave it as-is

closed 2021-03-10 by darraghbr