tetrabiodistributed/project-tetra #165: [Hardware] Build Progress Photos needed from UK Group

mmroden, 2021-03-04:
### Creating a task using this template
1. Rename Task title
2. Add details of task -- Darragh has built some splitters, need progress
3. Add reference of any existing Issue
4. Expand checklist below (if needed)
5. Add/Create appropriate label(s)
6. Select Project
7. Add/Create Milestone
8. Add any Assignee(s)

Task details
#(related issue number)

- [ X] Assigned _(specify assignees)_
- [ ] Work in Progress _(move card on Project Board)_
- [ ] Submit for Review _(assign reviewer, comment, & move on Project Board)_
- [ ] Working on Review changes _(comment w/ list of changes and date)_
- [ ] Review complete _(assigned reviewer comment with approved date)_

Judoguy12, 2021-08-22:
Testing pushfit adaptors
[image][image][image][image]

Judoguy12, 2021-08-22:
Testing 12mm pressure drop
[image][image]

Judoguy12, 2021-08-22:
[image][image]

mmroden, 2021-08-23:
Instead of tape, how about an o-ring?