tetrabiodistributed/project-tetra #168: [Hardware] pushfit adapters to 22m

mmroden, 2021-05-15:
### Creating a task using this template
1. Rename Task title
2. Add details of task
3. Add reference of any existing Issue
4. Expand checklist below (if needed)
5. Add/Create appropriate label(s)
6. Select Project
7. Add/Create Milestone
8. Add any Assignee(s)

Task details
#(related issue number)

- [ ] Assigned _(specify assignees)_
- [ ] Work in Progress _(move card on Project Board)_
- [ ] Submit for Review _(assign reviewer, comment, & move on Project Board)_
- [ ] Working on Review changes _(comment w/ list of changes and date)_
- [ ] Review complete _(assigned reviewer comment with approved date)_

Judoguy12, 2021-08-21:
I have just tested plastic threaded adaptors and they seem to fit together with the threaded part well now Ill move onto testing pressure drop across the pipe

Judoguy12, 2021-08-21:
[image]

PR #175 (2021-08-22): Upload printable push fit adapters
These adapters are compatible with 3/8" BSPT fittings and because Fusion 360 cannot model tapered threads were made by importing a part with 3/8" BSPT threads from McMaster-Carr, scaling them up in the radial direction by 0.5% and performing a boolean operation to cut the scaled up threads into the parts.

Closes #168
Closes #169
closed 2021-08-22 by Judoguy12