tetrabiodistributed/project-tetra #170: [Hardware] build up working tetra electronics with sensirion and pi board

mmroden, 2021-05-15:
### Creating a task using this template
1. Rename Task title
2. Add details of task
3. Add reference of any existing Issue
4. Expand checklist below (if needed)
5. Add/Create appropriate label(s)
6. Select Project
7. Add/Create Milestone
8. Add any Assignee(s)

Task details
#(related issue number)

- [ ] Assigned _(specify assignees)_
- [ ] Work in Progress _(move card on Project Board)_
- [ ] Submit for Review _(assign reviewer, comment, & move on Project Board)_
- [ ] Working on Review changes _(comment w/ list of changes and date)_
- [ ] Review complete _(assigned reviewer comment with approved date)_

mechatroniks-git, 2021-05-16:
Board built, wired to sensors, power tested with Raspberry Pi 4. Ready for Raspberry Pi 3B+ and Pi Display.

mmroden, 2021-08-11:
To be done in the US

mmroden, 2021-08-14:
Blocked on the UK prototype because we can't test it without the prototype