tetrabiodistributed/project-tetra #172: [Hardware] Moving to 12mm Push-fit

Judoguy12, 2021-07-09:
### Creating a task using this template
Having tested 6mm push-fit the pressure drop was found to be much too high and not enough flow was reaching the lung. To alleviate this we are moving to using 12mm push-fit piping. This means new 22mm adaptors are needed for the larger size

Task details
#171 

- [ ] Assigned @darraghbr 
- [ ] Work in Progress _(move card on Project Board)_
- [ ] Submit for Review _(assign reviewer, comment, & move on Project Board)_
- [ ] Working on Review changes _(comment w/ list of changes and date)_
- [ ] Review complete

darraghbr, 2021-07-11:
Can we have some supporting information here about the move to 12mm, for instance an image of the test set up with the 6mm system in situ showing that it is unable to crack the non-return valves. Which non-return valves did we test in the end? Was it just the John Guest stuff? If so I might get some SMC ones just in case they are much better.

Judoguy12, 2021-08-22:
Pressure Drop Across 30cm of 12mm tube was undetectable with the tools I have available. I tested at both 19cmH2O and 60cmH2O. The images below show the testing Setup.
[image]

19cmH2O
[image]
[image]

60cmH2O
[image]
[image]

mmroden, 2021-08-22:
Have you replicated this capability?

https://www.markmroden.com/Tetra-Testing/24-April-2020-ventilator-splitting

That is, shown that one patient can have a different PIP than the other?

darraghbr, 2021-08-22:
Not yet Mark, this is in direct response to my request in the comment above Jamies.

mmroden, 2021-08-22:
Gotcha, makes sense.

Judoguy12, 2021-08-28:
Whole circuit test setup at 45cm H2O

Found that lungs deflated very slowly and need a thinner one way valve to work
[image]
[image]

Judoguy12, 2021-08-28:
[image][image]

Judoguy12, 2021-08-28:
[image][image]

Judoguy12, 2021-08-28:
With a weaker one way valve operation would be much more optimal and lungs would deflate at the normal rate. After rearranging the inspiratory arm to the correct configuration the flow restriction worked much better going to 24.4 when blue is 2 turns down with pink remaining unaffected at 25.7

Judoguy12, 2021-09-12:
With more compliant valves on both inspiratory and expiratory of breathing circuits we get pressures as in the vent except a 1cmH2O drop in PEEP
[image]

Judoguy12, 2021-09-12:
[image]

Judoguy12, 2021-09-12:
[image]

Judoguy12, 2021-09-12:
With blue shut off and a compliant valve from there used in pink we notice no change to PIP but the PEEP does get much closer to the 9.5 so the bias circuit needs a soft valve too[image]

Judoguy12, 2021-09-12:
[image]

Judoguy12, 2021-09-12:
[image]

Judoguy12, 2021-09-12:
Also upon significantly reducing blues flow valve we found pink's PIP and PEEP not to change. I believe this is the requirement asked for by @mmroden .[image]

Judoguy12, 2021-09-12:
[image][image]

PR #171 (2021-07-10): Upload 12 mm push fit parts for manufacture
NB I have only made changes to the connector, no changes to the profile of the adapters has been made thus far.

Closes #172