tetrabiodistributed/project-tetra #176: [Hardware] pushfit adapters to 22mm leaking in FDM

Judoguy12, 2021-08-21:
### Creating a task using this template
1. Rename Task title
2. Add details of task
3. Add reference of any existing Issue
4. Expand checklist below (if needed)
5. Add/Create appropriate label(s)
6. Select Project
7. Add/Create Milestone
8. Add any Assignee(s)

Task details
#(related issue number)

- [ ] Assigned _(specify assignees)_
- [ ] Work in Progress _(move card on Project Board)_
- [ ] Submit for Review _(assign reviewer, comment, & move on Project Board)_
- [ ] Working on Review changes _(comment w/ list of changes and date)_
- [ ] Review complete _(assigned reviewer comment with approved date)_

Judoguy12, 2021-08-21:
The Current 22mm Adaptors are leaking when Printed using FDM. They leak at 19cmH2O. This is likely due to the relatively short taper and also the grooved nature of the FDM prints.  Tomorrow I will wrap the connector in tape to seal the mating surface and retest .If Printed in Resin or machined I do not believe the leak would be an issue. In the Photo the connector is connected to the inspirtory arm of the ventilator and then to a check valve opposing the flow of air to create a seal. It was tested in a soapy water mix. Images of the Test  carried out today where they are leaking are attached. If Printed in Resin or machined I do not believe the leak would be an issue. 

[image]
[image]

Judoguy12, 2021-08-22:
Once I wrapped black tape around I retested under same conditions and got a much smaller negligible leak[image]

mmroden, 2021-08-24:
Should be testing leak between 80 and 120 cmH20 of pressure.

Also, investigate an o-ring solution vs tape

mmroden, 2021-08-24:
We investigated a lot of leak before, the resin prints may help, but a proper ISO 5356 taper should also do the trick, esp with an o-ring on it.

mmroden, 2021-08-24:
80-120 is the standard per Daniel for pressures in the hospital, so 120 is a good target.  No need to go above that.

mmroden, 2021-08-24:
Also, a pre-use check will be useful for making sure that everything's good.  This is a circuit-safety check and device safety check prior to being put on a human.  Check the sparrow preuse checks on the puritan bennet ventilator in the EUA documentation (not the standard, but will work for EUA).