tetrabiodistributed/project-tetra #181: [Hardware] 3D printed adapters showing signs of leakage

darraghbr, 2021-10-03:
A significant leak has been observed on the current iteration of the push fit system.

It is suspected that it is coming from an FDM printed adapter between the push fit parts and the VentMon.

We need to identify the leaks exact location and remediate as we go. 

@Judoguy12 please use this issue to document where you locate the leaks and we will go from there.

I have pre-emptively redesigned the Sensirion adapters so they will effectively be an intereferance fit on the Sensirions and have printed some in a flexible resin (Photocentric Flexible LCD) these appear to seal to the Sensirions very well (so much so that I can fire the PCB clean off if I dead head it and blow hard into it)

[image]

In this image the purple is our adapter, and the cyan is the Sensirion.

I will be publishing these files in a new branch but keeping the PR as a draft.

darraghbr, 2021-10-04:
Honestly using the Sensirion I have and printing the adapters as they are on the master branch in some Photocentric flex resin they seal perfectly well, I can still blow the PCB clean off the sensor if I blow hard enough. So I don't think the #182 PR will be necessary.

Judoguy12, 2021-10-21:
Solved by the resin adaptors[image]